kevincar/bless #25: Supporting multiple central devices

cspensky, 2021-03-01:
While the BLE specification supports having multiple centrals connected to a single peripheral, Bless does not seem to support this functionality.  Indeed, it appears that BlueZ also inherently does not support this functionality.

I am currently considering implementing some functionality to circumvent BlueZ to enable this functionality and re-writing the code to support this type of interaction.  @kevincar Do you have any thoughts or ideas on this before I proceed?

There a lot of instances where this type of interaction makes sense in my opinion.  For example, multiple phones may want to connect to an IoT device (e.g., a light or a lock) and manipulate it.

kevincar, 2021-03-02:
Thank you @cspensky for bringing this up! I agree, without support for multiple central devices, this package remains limited. I  recently tested multiple devices from a macOS peripheral to two iPhone centrals and the package was working fine, albeit it was simply checking read/write/notify capabilities simultaneously. There are certainly platforms and scenarios where I haven't tested though.

I wasn't aware of this limitation of BlueZ, so I haven't thought much about this problem. If you have a working implementation for any lacking functionality, and if you've already thought up a circumventing implementation for BlueZ, pull request are very welcome.

Importantly, I think adding a test to our hardware tests to validate that 2+ centrals can interact with the peripheral would help here as well.

cspensky, 2021-03-03:
I'm happy to hear that it works well on macOS.  I found a library that seems to handle it properly on Linux using dbus (https://github.com/Douglas6/cputemp).  I'll investigate and try to fold what they are doing into Bless.  There will likely be a large merge request coming from me soon.

kevincar, 2022-05-28:
@cspensky 

I'm currently subscribing to a peripheral using bless running on Ubuntu and subscribed central devices are all getting the updates. 

If you're still seeing this let me know. Otherwise, closing this for now.

closed 2022-05-28 by kevincar