kevincar/bless #61: KeyError: 'A07498CA-AD5B-474E-940D-16F1FBE7E8CD' in server.py

WouterJD, 2022-01-10:
**The problem**
KeyError: 'A07498CA-AD5B-474E-940D-16F1FBE7E8CD'

**Reproduction**
On windows 10: `..\examples\server.py` 

**Expected behavior**
I would like to get acquanted with bless and hence try to run server.py, expecting it would run.

**Screenshots**
```
Z:\Github\kevincar\bless\wd-StartUp>..\examples\server.py
DEBUG:asyncio:Using proactor: IocpProactor
DEBUG:bless.backends.winrt.server:Creating a new service with uuid: A07498CA-AD5B-474E-940D-16F1FBE7E8CD
DEBUG:bless.backends.winrt.server:Adding service to server with uuid A07498CA-AD5B-474E-940D-16F1FBE7E8CD
Traceback (most recent call last):
  File "Z:\Github\kevincar\bless\examples\server.py", line 93, in <module>
    loop.run_until_complete(run(loop))
  File "C:\Users\Wouter Dubbeldam\AppData\Local\Programs\Python\Python38\lib\asyncio\base_events.py", line 612, in run_until_complete
    return future.result()
  File "Z:\Github\kevincar\bless\examples\server.py", line 67, in run
    await server.add_new_characteristic(
  File "C:\Users\Wouter Dubbeldam\AppData\Local\Programs\Python\Python38\lib\site-packages\bless\backends\winrt\server.py", line 196, in add_new_characteristic
    service: BlessGATTServiceWinRT = self.services[service_uuid]
KeyError: 'A07498CA-AD5B-474E-940D-16F1FBE7E8CD'

```
**Desktop (please complete the following information):**
 - OS: Windows 10

**Additional context**
No additional information, using server.py out of the box.

My name is Wouter Dubbeldam from the Netherlands and I am delighted that there is a library to create BLE-servers; it might be a great addition to FortiusAnt.
Pointing me in the right direction would be really appreciated!

WouterJD, 2022-01-11:
Hmmmm...

Changing
```
    my_service_uuid = "A07498CA-AD5B-474E-940D-16F1FBE7E8CD"
    my_char_uuid = "51FF12BB-3ED8-46E5-B4F9-D64E2FEC021B"
```
into
```
    my_service_uuid = "a07498ca-ad5b-474e-940d-16f1fbe7e8cd"
    my_char_uuid = "51ff12bb-3ed8-46e5-b4f9-d64e2fec021b"
```

solves the problem....

(Not that examples\server.py now works, but that may be another issue)

WouterJD, 2022-01-11:
Perhaps **def add_new_characteristic** 
should add `service_uuid = str(UUID(service_uuid))` 

like **def update_value** does

kevincar, 2022-01-11:
Thanks @WouterJD!

Feel free to make a pull request for this if this solves the issue, otherwise I’ll see if I can get to this soon.

WouterJD, 2022-01-11:
Will do.
Thanks for reaction, struggling on windows 10...

kevincar, 2022-01-17:
closing this since it has been merged to the develop branch

closed 2022-01-17 by kevincar