kevincar/bless #69: Bless fails to launch on Raspberry Pi0 because of some security issue

WouterJD, 2022-01-18:
**The problem**
Bless fails to launch on Linux with the error: txdbus.error.RemoteError: org.freedesktop.DBus.Error.AccessDenied: Connection ":1.136" is not allowed to own the service "org.bluez.TestService" due to security policies in the configuration file

**Reproduction**
Run an example script: server.py.

**Expected behavior**
Working bless on Linux..

**Screenshots**
See: https://pythonissues.com/issues/1750311

**Desktop (please complete the following information):**
 - OS: Raspberry Pi zero W

**Additional context**
Searching for the problem I got on pythonissues; copied here for reference - and of course an answer if there is any

kevincar, 2022-01-18:
Closing because this is a duplicate of #60

closed 2022-01-18 by kevincar

WouterJD, 2022-01-18:
sorry, overlooked that one