kevincar/bless #71: NSConcreteMutableData being sent instead of bytes

beachcitiessoftware, 2022-02-17:
**The problem**
Provide a clear and concise description of what the problem is.

I've created a basic GattServer on Mac, and instead of receiving bytes, or bytearray in the write_request_func.

**Reproduction**
Provide instructions or code samples that demonstrate the issue if possible>
```
    def _write(self, characteristic: BlessGATTCharacteristic, value: Any, **kwargs):
        characteristic.value = value
        if isinstance(value, bytearray) or isinstance(value, bytes):
            logger.debug(f'Char. {characteristic.uuid}, value set to: {value.hex().upper()}')  #  This never hits
```

**Expected behavior**
Provide a clear and concise description of what you expected to happen.
It would be preferable to receive a device agnostic value, or at least provide a way to get from NSConcreteMutableData to bytes

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows, macOS, Linux]

**Additional context**
Add any other context about the problem here.

beachcitiessoftware, 2022-02-17:
n/m value.__str__() has the desired effect.

kevincar, 2022-02-18:
Thanks for this. I think this is helpful to know since it still may be worth while to add some checks to still validate os agnostic values here. Will look into this.