kevincar/bless #73: How to use method "self.BlessServer.is_connected()"

WouterJD, 2022-03-03:
**The problem**
How to use method `BlessServer.is_connected() `

**Reproduction**
The following statement is not allowed (assignment in wait):
    `await b = self.BlessServer.is_connected()`

**Expected behavior**
    `b = len(self.BlessServer._subscribed_clients) > 0`

**Suggestion**
Perhaps the method should not be async, but I'm not very familiar with asynchronous coding.

kevincar, 2022-03-04:
I would recommend perusing the the [async documentation](https://docs.python.org/3/library/asyncio.html). There are some short, simple examples of implementation.

Perhaps you meant to use: `b = await self.BlessServer.is_connected()`

I should also acknowledge that the `is_connected` function is currently a misnomer. Not all backends (Windows/macOS/DBus) check for connectivity in the same way. The way I've implemented `bless` to check for connectivity is to determine whether a peripheral device has subscribed to any subscribable characteristics. As a result `is_connected()` currently only returns `True` when the client has subscribed to an available characteristic that is broadcast by `bless`

WouterJD, 2022-03-04:
Thanks, sorry for not finding this in docu.
The IsConnected() works as "desired" ; when the client aborts, it returns False

closed 2022-03-04 by WouterJD

kevincar, 2022-03-04:
No problem