kevincar/bless #74: Would you be interested in a FTMS_gatt_server as an example?

WouterJD, 2022-03-03:
**Describe the solution you'd like**
Gatt_server.py is a good start to create a server and I managed to create an FitnessMachineServer.

**Describe alternatives you've considered**
If you like, I can create a sample code (removing all application dedicated stuff) for reference.

**Additional context**
This example can be added to the examples

kevincar, 2022-03-04:
Possibly. Can you describe the purpose it would serve?

kevincar, 2022-03-04:
Sounds good. Make a pull request on the issue.

kevincar, 2022-05-31:
See [here](https://github.com/kevincar/bless/pull/77#issuecomment-1142577057)

closed 2022-05-31 by kevincar

WouterJD, 2022-05-31:
Hi Kevin

 

I know that this issue is closed.

 

My question is when you finalize your development and merge into master, so that I can use the master branch as a reference.

Thanks, Wouter

kevincar, 2022-06-01:
Merge and version update should happen in the next couple days. Just wanted to get some of the other issues out of the way.

WouterJD, 2022-06-01:
👍Please keep me posted

kevincar, 2022-06-08:
@WouterJD

[New version](https://github.com/kevincar/bless/releases/tag/v0.2.4) is released

WouterJD, 2022-08-09:
Thanks, I published BLE/bless version - using your published version.

WouterJD, 2022-10-11:
👍Met vriendelijke groetWouter Dubbeldam -------- Oorspronkelijk bericht --------Van: Kevin Davis ***@***.***>Datum: wo 8 jun. 2022 16:11Aan: kevincar/bless ***@***.***>Cc: Wouter Dubbeldam ***@***.***>, Mention ***@***.***>Onderwerp: Re: [kevincar/bless] Would you be interested in a FTMS_gatt_server as an example? (Issue #74)
@WouterJD
New version is released