kevincar/bless #75: QUESTION: How to defined the Generic Access Profile (during BlessServer creation)?

WouterJD, 2022-03-14:
* bless version: current
* Python version: 3
* Operating System: Windows 10, Raspberry

### Description

`BlessServer(name=myServiceName, loop=self.loop)` creates the BlessServer;
application services are added with `self.BlessServer.add_gatt(myGattDefinition)`

### What is already working

The service is created and can be inspected (using bleak) and the Generic Access Profile is present (it's not in the myGattDefinition) so it appears to be created implicitly:
```
Service: 00001800-... (Handle: 1): Generic Access Profile
        Characteristic: handle=   2, uuid=00002a00-0000-1000-8000-00805f9b34fb props=['read'], value="rpiServer"
        Characteristic: handle=   4, uuid=00002a01-0000-1000-8000-00805f9b34fb props=['read'], value="00 00"
```

rpiServer is the HOSTNAME of the computer, not the name provided to the BlessServer-creation,

### What is NOT yet working --- QUESTION:
How to set the two characteristics:
```
DeviceNameUUID = '00002a00-0000-1000-8000-00805f9b34fb'
AppearanceUUID = '00002a01-0000-1000-8000-00805f9b34fb'
```
?

WouterJD, 2022-03-17:
Hi @kevincar could you have a look into this please?
It would be very helpful for me to understand how to set the two General Access Profile characteristics, so that my FTMS is publsihed witha correct name instead of the hostname where the FTMS is running.

Thanks for you support

kevincar, 2022-03-17:
Since this doesn't appear to be an issue just yet, let's move this to a discussion