kevincar/bless #78: GATT Server name no longer advertising correctly

Avstn1, 2022-04-21:
**The problem**
After updating to macOS Monterey 12.3.1 with an iPhone running iOS 15.4.2, the specified GATT Server name is no longer advertising and I'm given a default name of "Mac" whenever starting the Bless server. I've confirmed that it's advertising my data, it's just that the name is incorrect.

**Reproduction**
```
my_service_name = "server-test"
server = BlessServer(name=my_service_name, loop=loop)
```

**Expected behavior**
The server would advertise as "server-test".

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS Monterey 12.3.1

Crudough, 2022-05-02:
@kevincar I am also seeing this after updating to Monterey. @Avstn1 have you been able to find any solution?

Thanks!

Crudough, 2022-05-02:
I'l continue investigating as well -- thanks @kevincar appreciate the response!

kevincar, 2022-06-01:
This seems to be an issue with the `pyobjc` library. Just submitted an issue there. Will continue to tease apart the issue. 

I'll see if I can supply a temporary workaround that will need to rely on compiled code.

kevincar, 2022-06-04:
@Crudough, @Avstn1 

Can you confirm that you were seeing the local device name prior to Monterey? The backend of bless utilizes the `CoreBluetooth` framework which has a [limit on the amount of data that is advertised](https://developer.apple.com/documentation/corebluetooth/cbperipheralmanager/1393252-startadvertising?language=objc).  Since bless, advertises the primary service UUID by default, it may limit the size of the local name field. 

I'm assuming this wasn't a problem in the past given that you've created this issue but want to see if you can still see the local name if you shorten it to a name with < 10 bytes in length.

kevincar, 2022-06-06:
Tested this issue again using Objective-C and found that MacOS Mojave allows long local names to be advertised, but no longer in Monterey. This behavior seems to be a result of how apple has coded the `CoreBluetooth` framework.

I've currently implemented a "work around" in which service UUIDs are dropped from the advertisement when the local name is longer than 10 bytes.

If there are other possible implementations to be suggested, please reopen this issue with those suggestions, ideas, or references.

PR #81 (2022-06-06): 78 corebluetooth adv name
Resolves #78 

Sort of... Work around for how `CoreBluetooth` currently implements advertisement packets
closed 2022-06-06 by kevincar