kevincar/bless #83: Should the service name be advertised under Windows

paulhamsh, 2022-06-10:
**The problem**
my_service_name doesn't seem to show in LightBlue when using Windows - it shows as the Windows device name

**Reproduction**
    my_service_name = "Sparksy"
    server = BlessServer(name=my_service_name)

**Expected behavior**
Show as Sparksy

**Desktop (please complete the following information):**
 - OS: Windows 10, using tp-link external bluetooth dongle (UB500)

Is it supposed to work under Windows 10?

Jakeler, 2022-06-26:
No, the passed name is currently unused:
https://github.com/kevincar/bless/blob/v0.2.4/bless/backends/winrt/server.py#L64
Would have to check if it is possible to implement it with WinRT. 
It should work on Linux and Mac though.

kevincar, 2022-08-27:
As far as I can tell, [WinRT doesn't allow custom changes to `local_name` property](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.advertisement.bluetoothleadvertisementpublisher?view=winrt-22621) that is broadcast. I've implemented more of a workaround rather than a true solution, by manipulating the name of the Windows Bluetooth module device.

This results in changes to the name of the bluetooth device as it is listed in the Windows Registery Key, and thus requires that you run bless with elevated privileges **or** set user access to the specific bluetooth properties key that allows for name changes

Closing this for now with this work around. Merged into the develop branch for now with merge #89

closed 2022-08-27 by kevincar