kevincar/bless #88: Can't send custom advertisement

v1ack, 2022-07-24:
**The problem**
Provide a clear and concise description of what the problem is.

I want to make an advertisement with custom manufacturer data, but raises `dbus_next.errors.DBusError: Failed to parse advertisement.`

Without `ManufacturerData` everything works

**Reproduction**
Provide instructions or code samples that demonstrate the issue if possible>

NewAdv added to BlueZ campatibility (like in #85 )
```python
import asyncio
from asyncio import sleep

from bless.backends.bluezdbus.dbus.advertisement import BlueZLEAdvertisement, Type
from bless.backends.bluezdbus.server import BlessServerBlueZDBus
from dbus_next import Variant
from dbus_next.service import dbus_property

class NewAdv(BlueZLEAdvertisement):
    @dbus_property()
    def TxPower(self) -> "n":  # type: ignore # noqa: F821
        return self._tx_power

    @TxPower.setter  # type: ignore
    def TxPower(self, dbm: "n"):  # type: ignore # noqa: F821
        self._tx_power = dbm

async def main():
    server = BlessServerBlueZDBus(name="my_service_name")
    await server.setup_task

    await server.app.set_name(server.adapter, server.name)
    advertisement = NewAdv(Type.BROADCAST, 1, server.app)

    # ManufacturerData = {UINT16: Variant}
    advertisement.ManufacturerData = {0x0002: Variant("s", "test string")}
    advertisement.ServiceUUIDs = ["00002800-0000-1000-8000-00805f9b34fb"]

    advertisement._tx_power = -23
    server.app.advertisements = [advertisement]

    server.bus.export(advertisement.path, advertisement)

    iface = server.adapter.get_interface("org.bluez.LEAdvertisingManager1")
    await iface.call_register_advertisement(advertisement.path, {})  # type: ignore

    await sleep(60)

asyncio.new_event_loop().run_until_complete(main())
```

**Expected behavior**
Provide a clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04
 - bluez: 5.64

**Additional context**
Add any other context about the problem here.

v1ack, 2022-08-27:
I found an answer for my issue In the source code of bluez
```c
struct ad_data {
	uint8_t data[25];
	uint8_t len;
};

struct manufacturer_data {
	uint16_t id;
	struct ad_data data;
};
```
Variant type must be `ay`
So, `ManufacturerData = {2: Variant("ay", b'12')}` is valid
Looks like it should be mentioned in docs

kevincar, 2022-11-09:
@v1ack glad you found the solution. Since this seems to be DBus specific I'll close this. Please reopen if I'm missing something and you feel more should be implemented into bless.

closed 2022-11-09 by kevincar

copywrite, 2024-02-29:
> I found an answer for my issue In the source code of bluez
>
> ```c
> struct ad_data {
> 	uint8_t data[25];
> 	uint8_t len;
> };
>
> struct manufacturer_data {
> 	uint16_t id;
> 	struct ad_data data;
> };
> ```
>
> Variant [...]

I tried this on apple M1 pro but failed though