kevincar/bless #90: How to disconnect by bless

trueroad, 2022-09-06:
**Describe the solution you'd like**
I would like to disconnect by bless (i.e., by peripheral), not by central.

If I understand correctly,
`BlessServer.stop()` stops advertising but does not disconnect.
The destructor of `BlessServer` class also does not disconnect.
Therefore, unless disconnected by central, the connection keeps to remain even if the Python script that imports bless exits.

**Describe alternatives you've considered**
`BleakClient` class has `disconnect()` method.
But, `BlessServer` class does not have the `disconnect()` method.

**Additional context**
I use Windows 10.

kevincar, 2022-11-09:
My understanding is that what you're describing is inherent to how Central and Peripheral devices are intended to function. Peripherals serve data to Central devices, while central devices connect to and read/write to peripheral data. Peripherals control advertising, while clients control the connection.

Because multiple clients could be connected at one time, it would be difficult for the peripheral to know whether it should disconnect all or just a specific one. Even if you got a handle on the device you wanted to disconnect, attempting to close the connection would depend on whether the device has no other app connections (See [here](https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.bluetoothledevice.close?view=winrt-22621))

If you truly need to stop all connections, you could consider shutting down the bluetooth module.

You could also consider implementing a protocol, where the `BlessServer` notifies the `BleakClient` of some command that then tells the `BleakClient` to disconnect.

Let me know if I'm way off here or if there is more to this than I'm understanding.

trueroad, 2022-11-13:
> Because multiple clients could be connected at one time, it would be difficult for the peripheral to know whether it should disconnect all or just a specific one. Even if you got a handle on the [...]

In my usage, the script by python-bless is the only application connected to the central device.
Is there a way to enumerate the devices connected in `BlessServer` ?
Furthermore, is there a way to call the `BluetoothLEDevice.Close` method in `BlessServer` ?

> If you truly need to stop all connections, you could consider shutting down the bluetooth module.

Is there a way to implement shutting down and re-start the Bluetooth module in Bless?

> You could also consider implementing a protocol, where the BlessServer notifies the BleakClient of some command that then tells the BleakClient to disconnect.

I use WIDI Master which is BLE-MIDI adapter as the central device.
It uses the BLE-MIDI protocol that has no disconnect command.

kevincar, 2022-11-14:
> Is there a way to enumerate the devices connected in BlessServer ?

At the moment, it cannot enumerate connected central devices

> is there a way to call the `BluetoothLEDevice.Close` method in `BlessServer`?

I'm assuming you're referring to [this method](https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.bluetoothledevice.close?view=winrt-22621) on Microsoft's Bluetooth API? [A quick search on their sample code](https://github.com/microsoft/Windows-universal-samples/search?q=BluetoothLEDevice) shows that this method is typically used by client (central) devices, not peripheral devices. 

So currently there isn't a way to call this from `Bless` because, to my knowledge, there's no clear way to call `BluetoothLEDevice.close` as a peripheral. Though I'm hoping to dive into this more.

> Is there a way to implement shutting down and re-start the Bluetooth module in Bless?

Currently, no. Though I'm open to exploring this.

> It uses the BLE-MIDI protocol that has no disconnect command.

Is your script attempting to control the standard bluetooth specification of the BLE-MIDI or are you attempting to use the BLE-MIDI specification to connect to your WIDI Master? From the [BLE-MIDI Specification](https://www.midi.org/specifications-old/item/bluetooth-le-midi), it looks like the WIDI functions as a bluetooth peripheral device (aka server) which broadcasts its information.

If you're attempting to connect to the WIDI over python, you should be using [Bleak](https://github.com/hbldh/bleak) as you pointed out earlier.

Since this doesn't seem to be a direct issue with bless. I'm converting this to a discussion so we can continue hashing out the problems you're having. If there really is a lack on the Bless side or a potential feature implementation, I'll open this back up as an issue.