kevincar/bless #92: server.py example: corebluetooth missing argument: 'max_write_without_response_size'

Kai0711er, 2022-10-03:
**The problem**
I am getting the following error when running the server.py example:
```
File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/bless/backends/corebluetooth/server.py", line 191, in add_new_characteristic
    await characteristic.init(service)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/bless/backends/corebluetooth/characteristic.py", line 78, in init
    super(BlessGATTCharacteristic, self).__init__(obj=cb_characteristic)
TypeError: __init__() missing 1 required positional argument: 'max_write_without_response_size'
```

**Reproduction**
Run the example code of server.py on macOS

**Expected behavior**
The example should run without errors...

**Screenshots**
None

**Desktop (please complete the following information):**
 - OS: macOS 12.6

**Additional context**
None

Thanks for the work on bless and let me know if I can provide any more information to resolve this issue

gegnew, 2022-10-13:
:point_up:

runman19, 2022-11-18:
Is there any workaround for this bug? E.g. by using previous versions of bless?

kevincar, 2022-11-18:
This is fixed in development. I should be merging this into the next version soon. If you need it to work now. You can install the current development version and it should work.

`pip instal git+https://github.com/kevincar/bless@develop`

runman19, 2022-11-19:
Just as a feedback: This development version seems to be working fine. I am still having some issues with the bidirectional data transfer, but this does not seem to be related to the bluetooth interface.

kevincar, 2022-11-19:
@runman19 Is the bidirectional data transfer related to this issue or something else?

GewoonGijs, 2022-12-29:
@kevincar, isn't a similar correction necessary for the other backends? It seems I got the same error on a RPi using de bluezdbus backend. Sorry, but I am don't understand the code well enough to make the correction myself.

@WouterJD, maybe you cn reflect on this as well.

closed 2023-07-01 by kevincar