kevincar/bless #93: How to know that a ble dongle is fully ble5 compliant?

WouterJD, 2022-10-22:
**The problem**
Function add_gatt() object has no attribute 'add-advertisement-status-changed'. 

I got this error on BLE4 dongles before, a user now reports it now occurs on his ble5 dongle. 
See 

**Reproduction**
Occurs always on same system

**Expected behavior**
How can I detect a dongle is really ble5 compliant;I recall there are inferior dongles that state compliancy but are not. 

**Screenshots**
See https://user-images.githubusercontent.com/69309725/197349798-2273ad49-6f32-438f-86e2-c39a27cebaa2.PNG

**Desktop (please complete the following information):**
 - OS: Windows

**Additional context**
I will ask user dongle details.

WouterJD, 2022-10-23:
A driver update solved the problem
See https://github.com/WouterJD/FortiusANT/issues/389#issuecomment-1288150709

The question (how to detect) remains open.

kevincar, 2022-11-09:
This may be driver dependent. One way to check on Windows is to checkout the device on device manager and confirm that the device driver specifies that "peripheral role is supported". 

Are you wanting to have bless implement a way to detect this?

WouterJD, 2022-11-10:
Hi @kevincar Thanks for response.
Yes, it would be very helpful and avoid issues and debugging to detect.

I can then implement the check and provide a decent message to the users.

kevincar, 2022-11-10:
I think this would be a nice feature. However, this may take some time as I'd like to ensure bless can work cross-platform, which would mean we'd ideally enable this across all platforms. 

In your case for Windows implementation, we should be able to accomplish this by [Enumerating appropriate device drivers](https://learn.microsoft.com/en-us/uwp/api/windows.devices.enumeration.deviceinformation.findallasync?view=winrt-22621#windows-devices-enumeration-deviceinformation-findallasync(system-string)), then checking the PeripheralRoleSupported Property.

Linux should be straight forward, but macOS will likely pose the biggest challenge as obtaining information on available Bluetooth controllers (particularly external devices) is not so obvious.

We can create a feature branch of bless to get this going and slowly work on the other parts. Would be great to get your help on this, otherwise the turn around time may be a bit slow.

WouterJD, 2022-11-19:
The source of the issue was that some installation did not work (on raspberry) which first triggered the idea that the BLE interface would be incompatible (see https://github.com/WouterJD/FortiusANT/issues/402#issuecomment-1320887551).

Since that's not the case, `if "object has no attribute" in exceptionMsg: self.Message += "BLE-5 required." `  may be enough.

kevincar, 2022-11-19:
I'm actively looking into this issue (though slowly) since it may be helpful for us to enumerate devices and select specified bluetooth module if multiple are available ([see this issue](https://github.com/kevincar/bless/issues/86)). Currently, you can actually enumerate and select the adapter when using bless on Linux, but I'd like to extend that to the other platforms. So I'm hoping this feature will eventually be available. Currently still in the works for now.