kevincar/bless #98: Does bless have a versionID that can be retrieved?

WouterJD, 2022-11-18:
**Describe the solution you'd like**
Please support a **__version__** so that the version of the used library can be printed in a logfile
Of course any other option serving the purpose is OK.

**Describe alternatives you've considered**
n/a.

**Additional context**
print(bless.__version__) is helpfull when debugging

kevincar, 2022-11-18:
Rather than hardcoding the version number, which could introduce human error if the version number is not updated with each new iteration. Would something like this work for you?

```python
from importlib.metadata import version
version("bless")  # outputs: 0.2.4
```

WouterJD, 2022-11-19:
Hi @kevincar 

Thanks for this option; it sure works: `logfile.Write(s % ('bless',        importlib_metadata_version("bless") ))`

Do you have an explanation why this option works for numpy, but not the others?
```
       #logfile.Write(s % ('argparse',     importlib_metadata_version("argparse") ))
       #logfile.Write(s % ('binascii',     importlib_metadata_version("binascii") ))
       #logfile.Write(s % ('math',         importlib_metadata_version("math") ))
        logfile.Write(s % ('numpy',        importlib_metadata_version("numpy") ))
       #logfile.Write(s % ('os',           importlib_metadata_version("os") ))
```

WouterJD, 2022-11-19:
PS. `numpy.__version__` and `version("numpy")` both produce the same result (in my case 1.21.4).
I do not have the python (environment) knowledge to understand where the values come from and/or whether they are fundamentally identical.

kevincar, 2022-11-19:
The `version("<pkg_name>")` should work for install packages via `pip` or `conda` into your current python environment. Modules such as `ok`, `math`, `argparse`, etc are builtin modules that are part of the [Python Standard Library](https://docs.python.org/3/library/index.html), so you can roughly say that the version for `os` or `math` is the same as your current python version.

kevincar, 2022-11-19:
Closing this for now. Feel free to list ideas of this posted on is not sufficient.

closed 2022-11-19 by kevincar