kevincar/bless #99: add_advertisment_status_changed error on Windows for example server

Rhynorater, 2022-11-28:
**The problem**
When I run the server from the example, I get the below error:
```
PS C:\Users\Justin\Windows Scripts> python .\ble.py
C:\Users\Justin\Windows Scripts\ble.py:91: DeprecationWarning: There is no current event loop
  loop = asyncio.get_event_loop()
DEBUG:asyncio:Using proactor: IocpProactor
DEBUG:bless.backends.winrt.server:Creating a new service with uuid: A07498CA-AD5B-474E-940D-16F1FBE7E8CD
DEBUG:bless.backends.winrt.server:Adding service to server with uuid A07498CA-AD5B-474E-940D-16F1FBE7E8CD
Traceback (most recent call last):
  File "C:\Users\Justin\Windows Scripts\ble.py", line 92, in <module>
    loop.run_until_complete(run(loop))
  File "C:\Python\Python310\lib\asyncio\base_events.py", line 646, in run_until_complete
    return future.result()
  File "C:\Users\Justin\Windows Scripts\ble.py", line 53, in run
    await server.add_new_service(my_service_uuid)
  File "C:\Python\Python310\lib\site-packages\bless-0.2.4-py3.10.egg\bless\backends\winrt\server.py", line 167, in add_new_service
  File "C:\Python\Python310\lib\site-packages\bless-0.2.4-py3.10.egg\bless\backends\winrt\service.py", line 53, in init
AttributeError: 'NoneType' object has no attribute 'add_advertisement_status_changed'
```

**Reproduction**
Windows 11
Python 3.10.5
Run:
```
python -m pip install bless
python server.py
```

**Expected behavior**
It would run successfully.

Laura-VFA, 2023-09-01:
I have the same problem, did anyone solve it? I would appreciate it :)

kevincar, 2023-09-02:
@Laura-VFA Is this with the latest version too (v0.2.5)?

Laura-VFA, 2023-09-04:
Yes, I have the bless==0.2.5 version

Jnp95, 2024-08-27:
Same problem with bless==0.2.6 version 😞

Jnp95, 2024-08-27:
My Solution :

Think to **activate** the Bluetooth on Windows :
 
[image]