kevincar/bless #105: no mudule error pysetupdi

h-enes-simsek, 2023-03-19:
**The problem**

I tried to run the server.py and got the following error:
````
(VENV) redacted>python server.py
Traceback (most recent call last):
  File "redacted\Desktop\bluetoothTest\server.py", line 12, in <module>
    from bless import (  # type: ignore
  File "redacted\Desktop\bluetoothTest\bless\VENV\lib\site-packages\bless\__init__.py", line 41, in <module>
    from bless.backends.winrt.server import (  # noqa: F401
  File "redacted\Desktop\bluetoothTest\bless\VENV\lib\site-packages\bless\backends\winrt\server.py", line 20, in <module>
    from bless.backends.winrt.ble import BLEAdapter
  File "redacted\Desktop\bluetoothTest\bless\VENV\lib\site-packages\bless\backends\winrt\ble\__init__.py", line 1, in <module>
    from .adapter import BLEAdapter
  File "redacted\Desktop\bluetoothTest\bless\VENV\lib\site-packages\bless\backends\winrt\ble\adapter.py", line 5, in <module>
    from pysetupdi import devices  # type: ignore
```

**Reproduction**

I created a new venv
pip install bleak
pip install bless
python server.py

**Desktop (please complete the following information):**
Win10

**Additional context**
There is no lib called pysetupdi in PYPI, did you install it manually?

h-enes-simsek, 2023-03-20:
Thank you for your reply. I think it would be helpful to have your answer in your readme file.

kevincar, 2023-03-20:
Thanks for bringing this up. I'm going to keep this open for my records to remind me to update the code. As @h-enes-simsek pointed out, this should either be in the readme or ideally I'll update the setup.py to ensure this is installed during `pip install` on windows OS.

NotLawson, 2024-07-01:
If you run `pip install git+https://github.com/gwangyi/pysetupdi` it clones and installs in one command.

Chapoly1305, 2024-12-13:
> If you run `pip install git+https://github.com/gwangyi/pysetupdi` it clones and installs in one command.

Thank you, this helped a lot. Should be included in README

closed 2025-12-26 by kevincar