kevincar/bless #116: TypeError: tp_basicsize for type '_bleak_winrt_Windows_Foundation.EventRegistrationToken' (24) is too small for base '_winrt.Object' (32)

fernandoaestrella, 2023-10-30:
**The problem**
Provide a clear and concise description of what the problem is.

Copied the example from here https://github.com/kevincar/bless/blob/master/examples/server.py and ran it

**Reproduction**
Provide instructions or code samples that demonstrate the issue if possible>

Copied the example from here https://github.com/kevincar/bless/blob/master/examples/server.py to VS Code in a new python file, inside a virtual environment (virtual environment folder is called ".venv")

imported bless via 
python -m pip install bless
and ran it
an error indicated I didn't have bleak_winrt so i got it via
python -m pip install bleak-winrt
then I ran it again and got the error below

**Expected behavior**
Provide a clear and concise description of what you expected to happen.

Expected the server to run and advertise a BLE message

**Screenshots**
If applicable, add screenshots to help explain your problem.

Traceback (most recent call last):
  File "d:\dev\magic\py\sb\boxes\ble_s\main.py", line 12, in <module>
    from bless import (  # type: ignore
  File "d:\dev\magic\py\venv\Lib\site-packages\bless\__init__.py", line 41, in <module>
    from bless.backends.winrt.server import (  # noqa: F401
  File "d:\dev\magic\py\venv\Lib\site-packages\bless\backends\winrt\server.py", line 14, in <module>
    from bless.backends.winrt.service import BlessGATTServiceWinRT
  File "d:\dev\magic\py\venv\Lib\site-packages\bless\backends\winrt\service.py", line 4, in <module>
    from bleak_winrt.windows.devices.bluetooth.genericattributeprofile import (  # type: ignore # noqa: E501
  File "d:\dev\magic\py\venv\Lib\site-packages\bleak_winrt\windows\devices\bluetooth\__init__.py", line 10, in <module>
    import bleak_winrt.windows.devices.bluetooth.genericattributeprofile
  File "d:\dev\magic\py\venv\Lib\site-packages\bleak_winrt\windows\devices\bluetooth\genericattributeprofile\__init__.py", line 15, in <module>
    import bleak_winrt.windows.devices.enumeration
  File "d:\dev\magic\py\venv\Lib\site-packages\bleak_winrt\windows\devices\enumeration\__init__.py", line 15, in <module>
    import bleak_winrt.windows.foundation
  File "d:\dev\magic\py\venv\Lib\site-packages\bleak_winrt\windows\foundation\__init__.py", line 7, in <module>
    _ns_module = bleak_winrt._import_ns_module("Windows.Foundation")
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "d:\dev\magic\py\venv\Lib\site-packages\bleak_winrt\__init__.py", line 15, in _import_ns_module
    module = module_from_spec(spec)
             ^^^^^^^^^^^^^^^^^^^^^^
TypeError: tp_basicsize for type '_bleak_winrt_Windows_Foundation.EventRegistrationToken' (24) is too small for base '_winrt.Object' (32)

**Desktop (please complete the following information):**
 - OS: [e.g. Windows, macOS, Linux]
Win10

**Additional context**
Add any other context about the problem here.

magnusjjj, 2024-01-14:
Got the exakt same error, windows 11 :).

goodztc, 2024-05-20:
I encountered the same problem and found that using a lower version of Python can solve it, Verified on Python version 3.9.19

goodztc, 2024-05-20:
> Got the exakt same error, windows 11 :).
you can try Python version 3.9.19

LucasVos, 2024-08-21:
Same issue here.. I would love to stay on Python 3.12 instead of going down :-) 
Is there another workaround?

102-97-98-105, 2024-09-22:
If anyone else stumbles upon this topic: https://github.com/kevincar/bless/pull/129

fernandoaestrella, 2025-02-04:
Ok, I guess I'll close it. Thanks, everyone! First time doing this!

closed 2025-02-04 by fernandoaestrella