kevincar/bless #117: Setting name_overwrite=True causes permissions error

edcloudcycle, 2023-10-30:
Hi all,
Just trying out the library to see if I can use it to simulate one of our devices for testing. To do that it needs to pretend to be that device by having the same name, however setting `name_overwrite=True` causes a `PermissionError: [WinError 5] Access is denied`.

I can see there are a lot of other services available on the BLE, many from Windows providing mobile phone connection options. I guess those services don't want me changing the name. Maybe they need to be disabled or maybe I am better of with a USB BLE dongle that won't have anything else trying to use it.

All suggestions very welcome.

Thanks

Ed

edcloudcycle, 2023-10-30:
If I turn off Bluetooth in Windows I can see there is a radio_not_present error inside the service provider object so I need a more subtle tool. :)

kevincar, 2025-12-31:
Converting this to a discussion because it is not yet clear that this is an issue.