kevincar/bless #128: Connect multiple Clients to BLE Server

metinkale38, 2024-03-20:
Hello,

i implemented a server with bless successfully and it is working. I used the Basic Server Example.

The only problem is, that i can only connect with one device. While i am connected to the BLE-Server, other Devices cant see it anymore. I have to disconnect the first device to connect another.

I am not sure, whether this is a BT or Bless limitation, but it would be very helpfull, if there is a solution for that problem.

decodeais, 2024-03-20:
You are not alone, in this moment I made the same test. I changed the order. Only the first is working.

juliencouturier, 2024-05-07:
I have the same issue on linux but not on windows : I have to restart the bluetooth service and my script to be able to connect another client.

decodeais, 2024-05-07:
I checked the BLE source code and found somewhere in the script that only the first service is advertised. When I tried  to change it, I had no success. There seemed to be a limitation. May be it is really hardware dependend.

mklemarczyk, 2024-06-28:
Hello, I will be working on the multi-client scenario. I will check your problem.

tmcg0, 2024-09-19:
@mklemarczyk any luck on that?

kevincar, 2024-09-21:
@metinkale38 Thanks for bringing this up. Without diving into this yet, this issue seems to be a `bless` limitation more so than a hardware one. That said, we've definitely run into OS-specific interface challenges between windows, linux, and macOS. 

As of BLE 4.1, limitations on peripheral vs central was removed and BLE 5.0 no longer has any limitations. However, although the specifications do not limit this, this could be limited at the hardware level and even at the OS level if using a built in chip, which most systems come with.

Either way worth looking into.

mklemarczyk, 2024-10-03:
@tmcg0 No luck so far. I will need to acquire some more hardware for testing.
Currently to update you I am waiting for my new Raspberry Pi 5.

The often cause of multi-client not working is auto-disable of advertisement when first client is connected. I am digging into that.

tmcg0, 2024-10-04:
Awesome, will definitely keep an eye out for anything you think might be in the right direction. I took a look through the codebase but am not seeing anything obvious that would cause this. I think you're right that the root cause is the stop advertising after first client connect.

Weirdly enough for the project I'm working on I switched to using the python dbus bindings, and all seemed well for a bit, but now it seems this issue has emerged again with that implementation. Which makes me feel the issue is OS-level, and I'm not sure if there's an elegant cross-platform (if the issue is cross-platform) way to handle this. FWIW I'm on an Raspberry Pi 5 running their 64-bit Debian bookworm-based OS. 

I've found a few OS-level tweaks that seem to help, but not sure exactly which subsets of these operations are actually helping. And obviously still haven't fully tracked down whatever is stopping advertisement.

In case it helps anyone, here's some functions from my current codebase that do some of the OS-level operations:

```python3

def setup_bluetooth():
    """Setup the Bluetooth adapter"""
    if os.name == "posix" and os.uname().sysname == "Linux":
        try:
            subprocess.run(["sudo", "systemctl", "stop", "bluetooth"], check=True)
            subprocess.run(["sudo", "hciconfig", "hci0", "down"], check=True)
            subprocess.run(["sudo", "hciconfig", "hci0", "up"], check=True)

            if ensure_bluetooth_config():
                ble_logger.info("Bluetooth configuration is set up correctly.")

                # Restart the Bluetooth service
                subprocess.run(
                    ["sudo", "systemctl", "restart", "bluetooth"], check=True
                )

                # Wait for the Bluetooth service to fully start
                time.sleep(2)

                # Disable Bluetooth agent and set advertising
                subprocess.run(
                    ["sudo", "btmgmt", "-i", "hci0", "pairable", "off"], check=True
                )
                subprocess.run(
                    ["sudo", "btmgmt", "-i", "hci0", "connectable", "on"], check=True
                )
                subprocess.run(
                    ["sudo", "btmgmt", "-i", "hci0", "advertising", "on"], check=True
                )

                # Modify /etc/bluetooth/input.conf so that the device doesn't require input for pairing
                ensure_bluetooth_input_config()

                ble_logger.info("Bluetooth setup completed successfully.")
            else:
                ble_logger.warning("Failed to set up Bluetooth configuration.")
        except subprocess.CalledProcessError as e:
            ble_logger.error(f"Error during Bluetooth setup: {e}", exc_info=True)
    else:
        ble_logger.info(
            f"Bluetooth setup routine not implemented for this OS, {os.name}"
        )

def ensure_bluetooth_input_config():
    """Ensure the Bluetooth input configuration is set up correctly"""
    required_lines = ["IdleTimeout=0"]

    try:
        # If the file doesn't exist, create it
        if not BLE_INPUT_CONFIG_FILE.exists():
            subprocess.run(["sudo", "touch", str(BLE_INPUT_CONFIG_FILE)], check=True)

        # Copy the file to a temporary location
        subprocess.run(
            ["sudo", "cp", str(BLE_INPUT_CONFIG_FILE), str(BLE_TMP_INPUT_CONFIG_FILE)],
            check=True,
        )

        # Change permissions of the temporary file
        subprocess.run(
[... 151 lines cut ...]
        if policy is None:
            policy = ET.SubElement(root, "policy", group="bluetooth")

        # check if the allow element exists, if not add it
        allow = policy.find("./allow[@send_destination='org.bluez']")
        if allow is None:
            ET.SubElement(policy, "allow", send_destination="org.bluez")

        # write the updated content back to the temporary file
        tree.write(str(BLE_TMP_DBUS_CONFIG_FILE))

        # copy the temporary file back to the original location
        subprocess.run(
            ["sudo", "cp", str(BLE_TMP_DBUS_CONFIG_FILE), str(BLE_DBUS_CONFIG_FILE)],
            check=True,
        )

        # remove the temporary file
        BLE_TMP_DBUS_CONFIG_FILE.unlink()
        return True

    except Exception as e:
        ble_logger.error(f"Error updating D-Bus configuration: {str(e)}", exc_info=True)
        return False
```

Jakeler, 2024-12-07:
Surprisingly this completely works in my testing. I can connect with 2 clients and the data is forwarded as expected: when the server sends all clients receive the notify, also each of the clients can individually send to the server.

Tested on Arch Linux (default configs regarding bluez) and Windows. Cheap Bluetooth 5.0 dongle based on Realtek RTL8761B: https://linux-hardware.org/?id=usb:0bda-8771

Code https://github.com/Jakeler/ble-serial/pull/60: 
https://github.com/Jakeler/ble-serial/blob/server-mode/ble_serial/bluetooth/ble_server.py

I would even say it works too well right now. While there is some logging on client connect, as someone running a server you don't really know or control who is receiving or sending. The feature #114 with metadata would already help a lot. Also it would be good to have a method to disconnect specific clients and a configurable limit/filter.

mklemarczyk, 2024-12-27:
It might be caused by hardware restrictions. In my last tests oved December, I seem to crash my setup. I will continue testing on my older Rasp 3. @Jakeler what is the hardware you connect that BT dongle ? I am mostly testing inside a container that is linked via Dbus with Rasp-debian OS.