kevincar/bless #130: dbus_fast.errors.DBusError: Failed to register advertisement

badogyang, 2024-04-16:
**The problem**
  File "/usr/lib/python3.9/asyncio/base_events.py", line 647, in run_until_complete
  File "/usrdata/fanconn/opt/server.py", line 105, in run
    await server_args.start()
  File "/usr/lib/python3.9/site-packages/bless/backends/bluezdbus/server.py", line 90, in start
    await self.app.start_advertising(self.adapter)
  File "/usr/lib/python3.9/site-packages/bless/backends/bluezdbus/dbus/application.py", line 181, in start_advertising
    await iface.call_register_advertisement(  # type: ignore
  File "/usr/lib/python3.9/site-packages/dbus_fast/aio/proxy_object.py", line 108, in method_fn
    BaseProxyInterface._check_method_return(msg, intr_method.out_signature)
  File "/usr/lib/python3.9/site-packages/dbus_fast/proxy_object.py", line 80, in _check_method_return
    raise DBusError._from_message(msg)
dbus_fast.errors.DBusError: Failed to register advertisement

**Reproduction**
I used two modules of the same chip from different manufacturers, the difference lies in the Bluetooth version, one module is BT5.0, the other is BT4.2, is it necessary to be compatible when registering the broadcast

**Desktop (please complete the following information):**
 - OS: [Linux]

MoritzSpiller, 2024-09-22:
I observed the same issue on Ubuntu 24.04 LTS ( Kernel: Linux 6.8.0-44-generic, BlueZ: 5.72):

```
Traceback (most recent call last):
  File "/home/moritz/dev/ble_adv.py", line 122, in <module>
    loop.run_until_complete(start_provisioning(loop))
  File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/moritz/dev/ble_adv.py", line 97, in start_provisioning
    await server.start()
  File "/home/moritz/dev/.venv/lib/python3.12/site-packages/bless/backends/bluezdbus/server.py", line 90, in start
    await self.app.start_advertising(self.adapter)
  File "/home/moritz/dev/.venv/lib/python3.12/site-packages/bless/backends/bluezdbus/dbus/application.py", line 174, in start_advertising
    await iface.call_register_advertisement(advertisement.path, {})  # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/moritz/dev/.venv/lib/python3.12/site-packages/dbus_next/aio/proxy_object.py", line 92, in method_fn
    BaseProxyInterface._check_method_return(msg, intr_method.out_signature)
  File "/home/moritz/dev/.venv/lib/python3.12/site-packages/dbus_next/proxy_object.py", line 62, in _check_method_return
    raise DBusError._from_message(msg)
dbus_next.errors.DBusError: Failed to register advertisement
```

btmon provides this information, but so far, I'm not sure how to fix it:
```
> HCI Event: Command Complete (0x0e) plen 4                 #8 [hci0] 14.643911
      LE Set Extended Scan Response Data (0x08|0x0038) ncmd 1
        Status: Invalid HCI Command Parameters (0x12)
@ MGMT Event: Command Status (0x0002) plen 3          {0x0001} [hci0] 14.643972
      Add Extended Advertising Data (0x0055)
        Status: Invalid Parameters (0x0d)
bluetoothd[8518]: = src/advertising.c:add_client_complete() Faile..   14.644579
bluetoothd[8518]: @ MGMT Command:.. (0x0011) plen 16  {0x0001} [hci0] 14.673894
        UUID: Vendor specific
```

I will get back once I know more.

kevincar, 2026-01-13:
This error could result from a number of reasons; e.g., your system's `bluetoothd` daemon is off, security preferences are not aligned, or `bluetoothd` version is outdated.

Since you mention you were able to successfully advertise with a 5.0 dongle, this is more likely the culprit. Though peripheral support exists within the BLE specification, not all dongles/devices are implemented. In my experience, BLE 5.0-compatible dongles/devices most reliably implement peripheral role support.