kevincar/bless #137: Configurable connection and advertising parameters

Uvis-Acurable, 2024-09-05:
1. Currently Bless server does not allow configuring advertising packet data such as manufacturer data or data service data, however Core Bluetooth allows it.

2. Allow changing minimum and maximum connection interval values

oclyke, 2024-09-25:
Anyone working on this? I am interested in helping out.

rwkeane, 2025-01-17:
I have an in-progress change to support adding `manufacturer_data` - though it would be pretty easy for someone to add a follow-up to make `service_data` work as well (it would be pretty much copy-paste of my code), it's just not needed for my use case. Planning to put up a PR once I solve an ongoing issue related to my machine's Bluetooth adapter.

Will be Linux only for the moment (I don't have an Apple device to test with, but the code to make it work is only a few lines. Windows is too crazy for me to figure out), but I hope to circle back and add other OSs soon to expand support for the app I am writing.

Uvis-Acurable, 2025-07-07:
It should still be possible to update this. As an example:

```
let newAdvertisementData: [String: Any] = [
    CBAdvertisementDataLocalNameKey: "NewName",
    CBAdvertisementDataServiceUUIDsKey: [CBUUID(string: "180A")]
]

peripheralManager.startAdvertising(newAdvertisementData)
```

Uvis-Acurable, 2025-07-07:
OK, Apple has locked it, so no writes can be done. With BlueZ would work I suppose that would require rewriting the driver as it's based on Apples CoreBluetooth

kevincar, 2026-01-22:
This issue may be a duplicate of #101. 

There is now a `BlessAdvertismentData` object that can be passed to the `BlessServer.start()` function that handles all of this. As noted, each OS backend comes with its own restrictions on what values can actually be modified, with CoreBluetooth being the most restrictive and BlueZ the most flexible. During server start, each backend process this object for advertising accordingly.

I'm going to close this since #101 seems to address this, but please reopen if this is not true.