kevincar/bless #144: dbus_next.errors.DBusError: No valid service object found

rwkeane, 2025-01-18:
**The problem**
```
  File "/home/ryan/src/test2/test_server.py", line 16, in start
    await self.__server.start()
  File "/home/ryan/anaconda3/envs/test-bless2/lib/python3.11/site-packages/bless/backends/bluezdbus/server.py", line 87, in start
    await self.app.register(self.adapter)
  File "/home/ryan/anaconda3/envs/test-bless2/lib/python3.11/site-packages/bless/backends/bluezdbus/dbus/application.py", line 138, in register
    await iface.call_register_application(self.path, {})  # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ryan/anaconda3/envs/test-bless2/lib/python3.11/site-packages/dbus_next/aio/proxy_object.py", line 92, in method_fn
    BaseProxyInterface._check_method_return(msg, intr_method.out_signature)
  File "/home/ryan/anaconda3/envs/test-bless2/lib/python3.11/site-packages/dbus_next/proxy_object.py", line 62, in _check_method_return
    raise DBusError._from_message(msg)
dbus_next.errors.DBusError: No valid service object found
```

**Reproduction**
Run code as follows:
```
import asyncio
from bless import BlessServer

class Advertiser:
    def __init__(self):
        self.__server : BlessServer = None

    async def start(self, friendly_name : str):
        self.__server = BlessServer(friendly_name)
        await self.__server.start()

    async def stop(self):
        if self.__server is None:
            return
        await self.__server.stop()

if __name__ == "__main__":
    # Handle bluetooth stack.
    id = "20b9180c-555e-4d2a-83bb-6208323cf91b"
    advertiser = Advertiser()

    event_loop = asyncio.new_event_loop()
    event_loop.run_until_complete(advertiser.start("name"))
    event_loop.run_forever()
```

**Expected behavior**
Service should be advertised over python

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Ubuntu Linux (reproduced on both 22.04 and 24.04)
 - Code version: Both the version pulled by `pip` and the `main` branch of this repo.
 - Both when code is run as `sudo` and when run normally
 
**Additional context**
I spent hours on this. I've attached some of the files I've generated while trying to debug it, in case they help you more than they did me:

[dbus-monitor.txt](https://github.com/user-attachments/files/18462420/dbus-monitor.txt)

[service-status.txt](https://github.com/user-attachments/files/18462423/service-status.txt)

Please take a look when you can. Once I get this solved, I plan to push a PR up (it's just pending validation) to add `manufacturer_data` to the advertisement data

rwkeane, 2025-02-09:
@kevincar Please take a look if you have a minute

I need Bluetooth advertising capability for a research project I am working on, and am really at my wits end trying to figure out this issue

Thanks!

kevincar, 2025-02-09:
@rwkeane 

You can't initialize a server with `BleakServer.start()` until a GATT Tree is defined with at least a service and a characteristic object. Before calling `start()`, initialize a service on the server. This explains why the error is "No valid service object found", because no service object was ever added to the server before calling `start()`

Since this issue isn't a clear bug with Bless directly, I'll close this issue for now, but we can keep the discussion going here if needed.

closed 2025-02-09 by kevincar